magicbug/Cloudlog #2945: Clublog upload not working

stevendodd, 2024-01-30:
When I try the following URL `http://<IP address>/index.php/clublog/upload/` I get a blank page with no errors reported in the log:

INFO - 2024-01-30 10:53:05 --> Model "User_options_model" initialized
INFO - 2024-01-30 10:53:05 --> Controller Class Initialized
INFO - 2024-01-30 10:53:05 --> Model "Clublog_model" initialized
INFO - 2024-01-30 10:53:05 --> Config file loaded: /var/www/html/application/config/config.php
INFO - 2024-01-30 10:53:05 --> Helper loaded: file_helper
INFO - 2024-01-30 10:53:05 --> Final output sent to browser
INFO - 2024-01-30 10:53:05 --> Total execution time: 0.0070

If I try the real time option `http://<IP address>/index.php/clublog/realtime/M0SNZ` I get:

ERROR - 2024-01-30 10:53:35 --> 404 Page Not Found: Clublog/realtime

My station location is real time enabled and my username and password is set on my profile

magicbug, 2024-01-30:
/clublog/realtime doesn't exist only /index.php/clublog/upload the callsign after does not need to be provided anymore

stevendodd, 2024-01-30:
Thanks for the response, I just get a blank page with no upload file created and no response from that url: `http://<IP address>/index.php/clublog/upload`

stevendodd, 2024-01-30:
```
INFO - 2024-01-30 12:51:56 --> Config Class Initialized
INFO - 2024-01-30 12:51:56 --> Hooks Class Initialized
INFO - 2024-01-30 12:51:56 --> UTF-8 Support Enabled
INFO - 2024-01-30 12:51:56 --> Utf8 Class Initialized
INFO - 2024-01-30 12:51:56 --> URI Class Initialized
INFO - 2024-01-30 12:51:56 --> Router Class Initialized
INFO - 2024-01-30 12:51:56 --> Output Class Initialized
INFO - 2024-01-30 12:51:56 --> Security Class Initialized
INFO - 2024-01-30 12:51:56 --> Input Class Initialized
INFO - 2024-01-30 12:51:56 --> Language Class Initialized
INFO - 2024-01-30 12:51:56 --> Loader Class Initialized
INFO - 2024-01-30 12:51:56 --> Config file loaded: /var/www/html/application/config/cloudlog.php
INFO - 2024-01-30 12:51:56 --> Config file loaded: /var/www/html/application/config/bands.php
INFO - 2024-01-30 12:51:56 --> Config file loaded: /var/www/html/application/config/lotw.php
INFO - 2024-01-30 12:51:56 --> Helper loaded: url_helper
INFO - 2024-01-30 12:51:56 --> Helper loaded: security_helper
INFO - 2024-01-30 12:51:56 --> Helper loaded: language_helper
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/account_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/adif_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/admin_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/awards_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/contesting_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/eqsl_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/export_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/filter_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/general_words_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/gridsquares_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/lotw_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/menu_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/notes_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/options_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/qslcard_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/qso_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/station_lang.php
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/statistics_lang.php
INFO - 2024-01-30 12:51:56 --> Database Driver Class Initialized
INFO - 2024-01-30 12:51:56 --> Session: Class initialized using 'files' driver.
INFO - 2024-01-30 12:51:56 --> Migrations Class Initialized
INFO - 2024-01-30 12:51:56 --> Language file loaded: language/english/migration_lang.php
INFO - 2024-01-30 12:51:56 --> Database Forge Class Initialized
INFO - 2024-01-30 12:51:56 --> Model "Options_model" initialized
INFO - 2024-01-30 12:51:56 --> Model "User_options_model" initialized
INFO - 2024-01-30 12:51:56 --> Controller Class Initialized
INFO - 2024-01-30 12:51:56 --> Model "Clublog_model" initialized
INFO - 2024-01-30 12:51:56 --> Config file loaded: /var/www/html/application/config/config.php
INFO - 2024-01-30 12:51:56 --> Helper loaded: file_helper
INFO - 2024-01-30 12:51:56 --> Final output sent to browser
INFO - 2024-01-30 12:51:56 --> Total execution time: 0.0084
```

magicbug, 2024-01-31:
I'm guessing /uploads is writeable you'll know by going to Admin - Debug will give you any warnings.

stevendodd, 2024-01-31:
Yes it's writeable and there are no warnings

stevendodd, 2024-01-31:
[image]

stevendodd, 2024-01-31:
I think it's actually working… I just get no response

magicbug, 2024-02-01:
You should see qsos in Clublog if it is

stevendodd, 2024-02-03:
It's updating club blog as I say just no response (blank page) coming back from the url request

closed 2024-02-03 by magicbug