magicbug/Cloudlog #2947: PHP8.3: I do the betatesting and report warnings and errors

df8oe, 2024-01-30:
Everything I report is not an issue, but for improvement and adapting to PHP8.3:

Message: trim(): Passing null to parameter #1 ($string) of type string is deprecated
Filename: models/Logbook_model.php
Line Number: 297

df8oe, 2024-02-02:
It seems that you have done a very good job. This is the only warning I got up to now. Everything (I used) is working perfectly. I will test further (not a problem - I set the error_reporting for this domain to maximum verbosity...

magicbug, 2024-02-03:
Added a fix to the dev branch if you can let me know if it resolves it

df8oe, 2024-02-03:
Yes, it is working now. But allow me two remarks:
1) The commit is completely confusing because the file was formatted different. All code lines are marked as "modified" because of this. You loose the possibility to easy see the differences.
2) Another (simpler) solution would have been to cast the value: trim((string)$station['station_country']). If $station['station_country'] is undefined it would be set to an empty string - exactly what is intended (and worked before PHP8.2 flawlessly).

df8oe, 2024-02-03:
Perfect now - many thanks!

pa8s, 2024-05-07:
Any progress on this error? It is still present in version 2.6.11 (on PHP 8.3.6)

> Severity: 8192 
> Message: Creation of dynamic property CI_Cache::$file is deprecated 
> Filename: libraries/Driver.php 
> Line Number: 188

df8oe, 2024-05-08:
Now three additional issues popped up.

Steps to reproduce:
- log out
- log in, then you get

A PHP Error was encountered
Severity: 8192
Message: strlen(): Passing null to parameter #1 ($string) of type string is deprecated
Filename: libraries/Encryption.php
Line Number: 914

A PHP Error was encountered
Severity: Warning
Message: Cannot modify header information - headers already sent by (output started at /var/kunden/htdocs/selbst/cloudlog/system/core/Exceptions.php:283)
Filename: core/Input.php
Line Number: 351

A PHP Error was encountered
Severity: Warning
Message: Cannot modify header information - headers already sent by (output started at /var/kunden/htdocs/selbst/cloudlog/system/core/Exceptions.php:283)
Filename: helpers/url_helper.php
Line Number: 564

Now you can call index.php and you are logged in, so login process itself is working. But using the straight way you are not pushed to index.php.

df8oe, 2024-05-23:
Nothing changed until now, so I have added two small correcions. Too small for making a pull request, so I woll paste them here:

libraries/Encryption.php line 914:
`: strlen((string)$str);`

core/Input.php line 351:
`if(!headers_sent())
  setcookie($prefix.$name, $value, $expire, $path, $domain, $secure, $httponly);`

That fixes login issues.

commit f264e561 (2024-02-03): Should fix #2947