magicbug/Cloudlog #2948: Gridsquare Map default view shows all QSLs as confirmed

stevendodd, 2024-01-31:
Need to manually select confirmation checkboxes and click plot

[image]

[image]

phl0, 2024-01-31:
Also check settings in your profile for default QSL methods:

[image]

stevendodd, 2024-01-31:
@phl0 works great thanks

magicbug, 2024-01-31:
Merged into dev so marking as resolved unless someone shouts at me

closed 2024-01-31 by magicbug