magicbug/Cloudlog #2949: Public view gridsquares link shows empty map

stevendodd, 2024-01-31:
Clicking the link goes to the following URL which shows an empty map: https://stevendodd.ddns.net/cloudlog/index.php/visitor/satellites/M0SNZ

[image]

phl0, 2024-01-31:
Fix proposed in https://github.com/magicbug/Cloudlog/pull/2950.

stevendodd, 2024-01-31:
@phl0 I've manually tested this fix and it doesn't work

stevendodd, 2024-01-31:
@phl0 sorry my mistake it does work, should it be showing only satellites though?

stevendodd, 2024-01-31:
It also shows all as unconfirmed; perhaps a similar problem as https://github.com/magicbug/Cloudlog/issues/2948#issue-2109687414

phl0, 2024-01-31:
> @phl0 sorry my mistake it does work, should it be showing only satellites though?

Yepp as it says in the menu header.

stevendodd, 2024-01-31:
What I meant is why only satellites - the link on the public page says grid squares not satellite grid squares? Is there any configuration that allows you to show all work to get squares regardless of type.

The map is not empty but now it shows all as unconfirmed.

[image]

magicbug, 2024-01-31:
Its satellite grids as thats what i wanted at the time, ideally needs the interface from the login side so it can be filtered but hasn't got that far but its not a bug.

closed 2024-11-27 by magicbug