magicbug/Cloudlog #2970: A PHP Error was encountered, Awards - CQ Magazine WAZ

n5hc, 2024-02-13:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**A PHP Error was encountered**
Severity: Notice
Message: Undefined offset: 45
Filename: models/Cq.php
Line Number: 56

**A PHP Error was encountered**
Severity: Notice
Message: Undefined index: count
Filename: models/Cq.php
Line Number: 56

**To Reproduce**
Steps to reproduce the behaviour:
1. Click on 'Awards'
2. Click on 'CQ'

**Screenshots**
[image]
[image]
[image]
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [macOS 14.3.1 (23D60)]
 - Browser [safari ]
 - Version [17.3.1 (19617.2.4.11.12)]

n5hc, 2024-02-13:
Basically the ITU and CQ is swapped for JH5FTY (and a few other JA's) in QRZ.COM.

ITU should be 45
CQ should be 25

Perhaps some more validation?

phl0, 2024-02-14:
Fix proposed in https://github.com/magicbug/Cloudlog/pull/2971.

closed 2024-02-21 by magicbug