magicbug/Cloudlog #2974: Cloudlog Cat no frequency

Duffischer, 2024-02-15:
**Describe the bug**
I tried using Cloudlog Cat with my Icom IC7300. Cloudlog Cat shows the rig as online and also has the right mode. When I change the mode the mode is also changed in Cloudlog Cat. But no frequency is shown in Cloudlog Cat.

**To Reproduce**
1. Connect IC7300
2. Start Cloudlog Cat

**Expected behaviour**
I would expect the frequency to be updated as well as the mode.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 11
 - Browser Chrome
 - Version Cloudlog Cat 1.3.0.0

**Additional context**
Due to the fact that the mode is working I'm assuming that the missing frequency is an issue with the Cloudlog Cat and not the settings but please correct me if I'm wrong and just missed some setting.

magicbug, 2024-02-15:
Reinstall with https://github.com/magicbug/CloudlogCAT/releases/download/1.8/CloudlogCAT.Test.Release.ICOM.zip and it should resolve it

Duffischer, 2024-02-15:
> Reinstall with https://github.com/magicbug/CloudlogCAT/releases/download/1.8/CloudlogCAT.Test.Release.ICOM.zip and it should resolve it

Yep, frequency is now showing in the Cloudlog Cat but now the sync to my Cloudlog isn't working anymore. I added the URL and the API key, but it doesn't sync.

Duffischer, 2024-03-17:
@magicbug Do you have any update for me on this issue?

Reset4350, 2024-08-13:
Hi -- I just installed v1.8 today and having the same issue. I now have the frequency from my 7300, but getting, "Cloudlog Synced: Failed". I see packets flowing between the client and the server, but the client does not seem to be happy. :)

[image]

[image]

Reset4350, 2024-08-21:
Just to follow up on my experience, I dug around the interwebs and found if I use the following URL/ API for CloudLog things start working. The URL/API used is not what shows up in Cloudlog to copy and paste into the app using the API key: 

[image]

Duffischer, 2024-08-21:
@Reset4350 
I in fact found out that a new version was released at some point which did work with my Icom and the sync was working as well. 

Just tried to look up the exact version but sadly the info page in the cloudlog cat shows just 1.0.0.0.
I did also clear my downloads folder recently so I was not able to retrieve the version number from there. 

Did you download the newest version from the Cat project?

Reset4350, 2024-08-21:
I downloaded this version of CloudLog Cat and it cured the frequency issue with the 7300. It shows as version 1.8. I don't think that the changes in 1.8 have been incorporated in the latest stable release of CloudLog Cat: https://github.com/magicbug/CloudlogCAT/releases/download/1.8/CloudlogCAT.Test.Release.ICOM.zip

closed 2024-09-27 by magicbug