magicbug/Cloudlog #2983: Error in RX_PWR value in ADIF does not throw error in ADIF import

Reid-n0rc, 2024-02-21:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
I had an ADIF file with several invalid values for the RX_PWR field in the ADIF file. I had entered the cut number in the ARRL DX CW contest so the field ended up with values like `KW` and `1TT` instead of `1000` and `100`. This resulted in an error and the QSOs not being added to Cloudlog. When the import was complete, the `Yay, its imported!` message was shown with no errors. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'ADIF Import/Export'
2. click on 'Choose File.' Select a ADIF file with invalid values in the `RX_PWR` field.
3. Click on 'Upload'
4. See no error message after `Yay, its imported!`
5. Check the log for imported QSOs
6. See imported QSOs are not in the log.

**Expected behaviour**
When the ADIF file has an error that prevents the qso from being imported, there should be an error shown in the upload result.

magicbug, 2024-02-21:
Oh, yes definitely need to put a filter on those as RX_PWR fields in ADIF are number only.. what apps putting that into those fields as that's incorrect

Reid-n0rc, 2024-02-21:
@magicbug I exported from N1MM

magicbug, 2024-02-21:
@Reid-n0rc could you drop me the file via email or on here be good to test against

Reid-n0rc, 2024-02-21:
Here ya go @magicbug 

[N0RC.adi.txt](https://github.com/magicbug/Cloudlog/files/14364408/N0RC.adi.txt)

magicbug, 2024-02-26:
Now throws an error with the reason and sets the field to null so it can import regardless

[image]

Hopefuly @Reid-n0rc this fixes your problem

Reid-n0rc, 2024-02-26:
Thank you @magicbug , you're the best!

commit aac251d2 (2024-02-26): [ADIF Import] Catch when RX_PWR is not a number and error out

If RX_POWER is not a number or K or KW then set the value to null and show error.

Fixes #2983
closed 2024-02-26 by magicbug