magicbug/Cloudlog #2992: Upcoming dxpeditions - error

spREGR, 2024-02-24:
Describe the bug
When attempting to use file_get_contents() to retrieve data from https://cdn.cloudlog.org/read_ng3k_dxped_list.php, a 403 Forbidden error is encountered.

To Reproduce
Steps to reproduce the behavior:

Use the PHP code snippet below:
php
```
$url = 'https://cdn.cloudlog.org/read_ng3k_dxped_list.php';
$data = file_get_contents($url);
```
Execute the PHP script.
Expected behavior
The file_get_contents() function should successfully retrieve the contents of https://cdn.cloudlog.org/read_ng3k_dxped_list.php without encountering a 403 Forbidden error.

Screenshots
[image]

Desktop: Windows 11

Cloud log hosted on OS: Debian 12 
Browser used to access from LAN: Microsoft Edge 
PHP Version: 8.2.7
Additional context

The URL https://cdn.cloudlog.org/read_ng3k_dxped_list.php is accessible via web browser without any issues.
This issue occurs consistently on different machines with different operating systems.
Attempting to set a custom user agent using stream_context_create did not resolve the issue.

magicbug, 2024-02-24:
On the server can you run `wget https://cdn.cloudlog.org/read_ng3k_dxped_list.php` and see if you get an output there

magicbug, 2024-02-24:
This could also be that allow_url_fopen is disabled in the php.ini

spREGR, 2024-02-26:
On the server, I get the forbidden message on wget command
```
root@lamp ~# wget -d https://cdn.cloudlog.org/read_ng3k_dxped_list.php
DEBUG output created by Wget 1.21.3 on linux-gnu.

Reading HSTS entries from /root/.wget-hsts
URI encoding = 'ANSI_X3.4-1968'
converted 'https://cdn.cloudlog.org/read_ng3k_dxped_list.php' (ANSI_X3.4-1968) -> 'https://cdn.cloudlog.org/read_ng3k_dxped_list.php' (UTF-8)
Converted file name 'read_ng3k_dxped_list.php' (UTF-8) -> 'read_ng3k_dxped_list.php' (ANSI_X3.4-1968)
--2024-02-26 15:23:19--  https://cdn.cloudlog.org/read_ng3k_dxped_list.php
Certificates loaded: 141
Resolving cdn.cloudlog.org (cdn.cloudlog.org)... 104.21.75.17, 172.67.166.123, 2606:4700:3036::ac43:a67b, ...
Caching cdn.cloudlog.org => 104.21.75.17 172.67.166.123 2606:4700:3036::ac43:a67b 2606:4700:3034::6815:4b11
Connecting to cdn.cloudlog.org (cdn.cloudlog.org)|104.21.75.17|:443... connected.
Created socket 3.
Releasing 0x00005556f8c53090 (new refcount 1).

---request begin---
GET /read_ng3k_dxped_list.php HTTP/1.1
Host: cdn.cloudlog.org
User-Agent: Wget/1.21.3
Accept: */*
Accept-Encoding: identity
Connection: Keep-Alive

---request end---
HTTP request sent, awaiting response...
---response begin---
HTTP/1.1 403 Forbidden
Date: Mon, 26 Feb 2024 15:23:20 GMT
Content-Type: text/html; charset=UTF-8
Transfer-Encoding: chunked
Connection: close
Accept-CH: Sec-CH-UA-Bitness, Sec-CH-UA-Arch, Sec-CH-UA-Full-Version, Sec-CH-UA-Mobile, Sec-CH-UA-Model, Sec-CH-UA-Platform-Version, Sec-CH-UA-Full-Version-List, Sec-CH-UA-Platform, Sec-CH-UA, UA-Bitness, UA-Arch, UA-Full-Version, UA-Mobile, UA-Model, UA-Platform-Version, UA-Platform, UA
Cross-Origin-Embedder-Policy: require-corp
Cross-Origin-Opener-Policy: same-origin
Cross-Origin-Resource-Policy: same-origin
Origin-Agent-Cluster: ?1
Permissions-Policy: accelerometer=(),autoplay=(),browsing-topics=(),camera=(),clipboard-read=(),clipboard-write=(),geolocation=(),gyroscope=(),hid=(),interest-cohort=(),magnetometer=(),microphone=(),payment=(),publickey-credentials-get=(),screen-wake-lock=(),serial=(),sync-xhr=(),usb=()
Referrer-Policy: same-origin
X-Frame-Options: SAMEORIGIN
cf-mitigated: challenge
Cache-Control: private, max-age=0, no-store, no-cache, must-revalidate, post-check=0, pre-check=0
Expires: Thu, 01 Jan 1970 00:00:01 GMT
Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=P3TR%2BEE0zbD8KBLP0oNGe1vhRGtlp06hnsamOTOUrn1yfPRJI%2BkF18GwAprZI8Hp9kHOb2JMFFf8%2BI9eA6f7ZHZsGGfXODdyCGwm8Wwb0brwmUZ6kchOnuWe7safYbufL1LR"}],"group":"cf-nel","max_age":604800}
NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
Vary: Accept-Encoding
Server: cloudflare
CF-RAY: 85b9304dee764325-EWR
alt-svc: h3=":443"; ma=86400

---response end---
403 Forbidden
URI content encoding = 'UTF-8'
2024-02-26 15:23:20 ERROR 403: Forbidden.
```

On both 
`/etc/php/8.2/apache2/php.ini`
and
`/etc/php/8.2/cli/php.ini`
I have 
`allow_url_fopen=On`

magicbug, 2024-02-26:
That helps looks like Cloudflare was blocking can you retry to see if that resolves it

spREGR, 2024-02-27:
Unfortunately, the cloudflare block persists. Is there a possibility of whitelisting the IP from VU land?

magicbug, 2024-03-13:
Apologies for the delay, been in contact with Cloudflare and hopefully now have something in place for India for some reason your ISPs IP block isn't trusted very well by quite a few providers.

If you can try it again and let me know the outcome

spREGR, 2024-03-18:
Thanks for looking into this. Unfortunately, its the same status - still blocked. I have two ISP (load balancing) - Airtel and Jio. They are India's biggest ISPs unlike small players many people use. Hope this helps

ncov20, 2024-03-26:
Unfortunately, access this URL from China meets the same problem. I also test two ISPs: China Mobile and China Unicom, both of them are China's biggest ISPs.

closed 2024-09-27 by magicbug