magicbug/Cloudlog #2996: Logbook overview - edit QSO menu is cutoff

LarsThunberg, 2024-03-03:
When edit a QSO in "Logbook" -> "Overview", the menu is cutoff. Best shown in the image below (in the red marked area):

**Expected behaviour**
Would it be possible to show the complete menu in this case?

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 11
 - Chrome latest version

magicbug, 2024-03-06:
Can you confirm which theme you have selected in your account settings

LarsThunberg, 2024-03-07:
It is the default team and the version of Cloudlog is 2.6.6

LarsThunberg, 2024-03-07:
Hmmmm.  I updated to 2.6.6 yesterday and now I can not see the original problem anymore:

[image]

magicbug, 2024-03-07:
Quite possible that it was a caching issue don't think I've touched any code in that area

closed 2024-03-08 by LarsThunberg