magicbug/Cloudlog #3016: Feature Request: Add support for saving SSTV images in a QSO

patrickrb, 2024-03-19:
- We should have the ability to save SSTV images that were part of a QSO
- I'd love to dive into working on this feature.  Wanted to post it here publicly since I'm new to the project and any feedback or guidance on best practices for this project would be appreciated.

Stories 
- Users should have an easy way to upload images from the live/post QSO pages
- Users should be able to view the associated images when viewing a logged SSTV QSO

I'll start a branch associated with this feature request/issue and keep this issue open to track progress.

patrickrb, 2024-03-19:
I'm going to start by looking at how EQSL cards are saved/loaded in this project and probably use that as s start for how I store the SSTV images.

phl0, 2024-03-19:
> I'm going to start by looking at how EQSL cards are saved/loaded in this project and probably use that as s start for how I store the SSTV images.

Better look at the QSL managements for (scans) of paper cards. eQSL images cannot be uploaded manually.

magicbug, 2024-03-19:
Yep seems best place, I'm wondering if it makes sense to have an "Additional Material" so that you can upload any file against a QSO so whether that's audio or an image that might better suit this overall task

patrickrb, 2024-03-25:
was able to get quite a bit of work done on this feature today.  all the writes are working, both the image data to a new directory `assets/sstvimages` and a new table `sstv_images`

[image]

I just need to work on showing the images that have been uploaded/saved, hoping to have it wrapped up for PR in the next week or so!

patrickrb, 2024-03-25:
> Yep seems best place, I'm wondering if it makes sense to have an "Additional Material" so that you can upload any file against a QSO so whether that's audio or an image that might better suit this overall task

I went ahead and started adding this specifically for SSTV, my plan is to only show the SSTV Management tab if the QSO mode is `SSTV` 

That said, I'd be happy to adjust my current PR for 'additional material', or even make another feature PR for audioQSL/additional material.

@magicbug let me know how youd like me to proceed.

PR #3031 (2024-03-28): Add SSTV image support
resolves #3016
closes #3016 

- Adds basic SSTV image support from viewing an existing QSO in the logbook
- If the QSO mode is `SSTV` then we show a new tab named "SSTV Management", otherwise we show nothing
  - New SSTV Management tab works similarly to QSL management tab, except we only have one input and it can handle multiple images at a time.
  - Users can delete or view an SSTV Image from the SSTV Management tab
- If there are any SSTV Images associated with the selected QSO then we show a "SSTV Images" tab
  - Inside the SSTV Images tab is a carousel similar to th [... cut]
closed 2024-04-01 by patrickrb