magicbug/Cloudlog #3039: Simple Fast Log Entry not entering correct default RST for FT8

Reid-n0rc, 2024-03-29:
**Describe the bug**
The simple fast log entry documentation states that for WSJT modes the default signal report should be -10. Cloudlog is setting this to 599. This is from the simple fast log entry documentation linked on the page and found here: https://df3cb.com/fle/documentation/

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'QSO > Simple Fast Log Entry
2. Enter `DATE 2024-03-27 17m 1403 3d2ajt ft8` in the 'Enter the Data' box and press enter.
3. 'QSO List' Shows RST(S) and RST(R) as 599

**Expected behaviour**
When the mode is a WSJT mode (like FT8), the default RST should be set to -10, not 599.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Firefox
 - Version 124.01

magicbug, 2024-03-29:
Will look into it next week that said there really shouldn't be a need to use FLE for ftx qsos

Reid-n0rc, 2024-03-29:
You're correct, there shouldn't be unless you're dumb like me, mess up the logging direct to Cloudlog, then accidentally delete the ADIF file so you're going back through your ALL.txt