magicbug/Cloudlog #3051: Feature request - Backup to cloud

patrickrb, 2024-04-04:
- As a server admin, id like to be able to back up the entire database to a cloud storage solution
  - Azure
  - AWS
  - Dropbox
  - Google Drive
  - Google cloud storage
- It feels like some config options in the admin section + a cron job would be sufficient
- Should be implemented so that each integration is separated into its own file for cleanliness  

I'll probably take a swing at this soon, any input/feedback/advice would in the meantime would be appreciated

pd5wl, 2024-05-25:
My be Add (S)FTP as well for those lucky enough to run there own systems (Sorry, comes with the job. Hi)

magicbug, 2024-11-27:
This really is a server function and not something Cloudlog needs to handle, there is plenty scripts available for backing up files to the cloud

closed 2024-11-27 by magicbug