magicbug/Cloudlog #3052: Post QSO not working correctly

ok1cdj, 2024-04-04:
Post QSO not working correctly when you not worked station before.
When you have callsing in log from previous time all is working.
When you trying to log a new station, the icon of QRZ is not shown and the data like DXCC continent are not populated.
You are not able to save QSO.  You need manualy select DXCC and then you able to save.
[image]

magicbug, 2024-04-04:
Weird, not seeing such a problem, have you tried recently to update the country files and its failed? if not i'd try that first

ok1cdj, 2024-04-04:
Yes, tried update country files with success.
It is happened only when SAT is selected.

INFO - 2024-04-04 10:47:16 --> Model "Logbook_model" initialized
INFO - 2024-04-04 10:47:16 --> Model "Logbooks_model" initialized
ERROR - 2024-04-04 10:47:16 --> Query error: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1 - Invalid query: select name, cqz, lat, 'long' from dxcc_entities where adif = 
ERROR - 2024-04-04 10:47:16 --> Severity: error --> Exception: Call to a member function result() on bool /var/www/hosts/log.sv0syh.eu/application/models/Logbook_model.php 4270
INFO - 2024-04-04 10:47:16 --> Session: Class initialized using 'files' driver.

magicbug, 2024-05-20:
I'm hopeful this has now been fixed

closed 2024-05-20 by magicbug