magicbug/Cloudlog #3062: PHP errors if SSTV dir does not exist

phl0, 2024-04-07:
PHP throws errors if the SSTV images directory does not exist and the user calls the view SSTV images menu item:

[image]

[image]

Maybe the dir should be created automagically on the first call.

patrickrb, 2024-04-07:
got a fix for this locally, trying to track down the issue with sstvimages directory not being added to the repo

phl0, 2024-04-08:
Fixed by https://github.com/magicbug/Cloudlog/pull/3063.

PR #3063 (2024-04-08): SSTV directory fixes
resolves #3062

This pull request primarily focuses on enhancing the file and directory handling in the application. The changes include an update to the `folderSize` function in `storage_helper.php` to handle cases where the input is not a directory or if the directory is missing, and the addition of an `index.html` file in `assets/sstvimages` to prevent directory access and also adds the `sstvimages` directory to the repo.

Here are the key changes:

File and Directory Handling:

* [`application/helpers/storage_helper.php`](diffhunk://#diff-413abc91cb9cb7f2cc0f661e201eea [... cut]
closed 2024-04-08 by phl0