magicbug/Cloudlog #3074: On upgrade QRZ icons & lookup not working for new stations

aaggiiss, 2024-04-13:
Upgraded my local version from 2.4.5 to v2.6.8 using the zip replacement.

I have found that my QRZ lookup does not work nor do the icons appear for new stations that I have not worked before which the previous version did.

When I roll back it works fine.

Previous version entering in a new callsign and then pressing tab the QRZ/HAM QTZ icon appears and and the details on the right, after upgrade I see nothing on pressing tab.

[image]

[image]

I have tried refreshing my browser cache (mac firefox) and also tried in in Safari  and my ipad 

Server Software 	Apache/2.4.6 (CentOS) PHP/7.4.33
PHP Version 	7.4.33
MySQL Version 	5.5.5-10.5.24-MariaDB

Browser. Firefox 124.0.2 (64-bit) 
MAC: 14.4.1 (23E224)

The user name and password for QRZ is correct in my config.php 

It does seem to work if I use a callsign that I have previously worked, so put in previously works callsign all works as expected, put in new callsign does not do the lookup or show the icons in either live or post QSO screens.

Any suggestions, many thanks 

Stuart

magicbug, 2024-04-17:
My first suggestion would be to turn on logging in /application/config/config.php and check for any errors in /application/logs

However you could also try <your-cloudlog-url>//index.php/logbook/json/2M0SQL/0/2m/FM/

That should do a lookup against the callbook you have setup in config.php and if you have still got "development" mode turned on in index.php you should see any errors that might be affecting things..

That would be my first steps

arvis-soldans, 2024-04-20:
same here....

> However you could also try //index.php/logbook/json/2M0SQL/0/2m/FM/

[image]

magicbug, 2024-04-20:
That sort of error makes me think of an issue with the country files

I'd quickly do a Admin -> Update Country files and let it run until it says done and see if the errors still there

arvis-soldans, 2024-04-20:
> That sort of error makes me think of an issue with the country files
> 
> I'd quickly do a Admin -> Update Country files and let it run until it says done and see if the errors still there

Thx... error from  photo solved, but now i get blank page.....  and don't get data from qrz.com xml :(

magicbug, 2024-04-20:
have you got a paid account on QRZ does changing it to HamQTH with a login make any difference etc

aaggiiss, 2024-04-28:
Hi, sorry for the delay in replying life, work etc - downloaded the latest version and tried again.

I am seeing

ERROR - 2024-04-28 15:54:56 --> Query error: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1 - Invalid query: select name, cqz, lat, 'long' from dxcc_entities where adif =
ERROR - 2024-04-28 15:58:21 --> Query error: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1 - Invalid query: select name, cqz, lat, 'long' from dxcc_entities where adif =
ERROR - 2024-04-28 16:00:29 --> Query error: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1 - Invalid query: select name, cqz, lat, 'long' from dxcc_entities where adif =
ERROR - 2024-04-28 16:01:21 --> 404 Page Not Found: Indexphp/logbook
ERROR - 2024-04-28 16:01:26 --> Query error: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1 - Invalid query: select name, cqz, lat, 'long' from dxcc_entities where adif =

So did the county updates a suggested after someone and ran your supplied URL and got the same.

A Database Error Occurred

Error Number: 1064

You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1

select name, cqz, lat, 'long' from dxcc_entities where adif =

Filename: models/Logbook_model.php

Line Number: 4306

Any thoughts?

Thanks

Stuart G7JYF

R1BLH, 2024-05-05:
Cloudlog 2.6.11 (dev), QRZ.com lookup when creating a new QSO still does not work...

```
<?php defined('BASEPATH') OR exit('No direct script access allowed'); ?>

ERROR - 2024-05-05 21:31:16 --> Severity: error --> Exception: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1 /var/www/r1blh.cats-home.net/log/system/database/drivers/mysqli/mysqli_driver.php 301

```

R1BLH, 2024-05-11:
> Cloudlog 2.6.11 (dev), QRZ.com lookup when creating a new QSO still does not work...
>
> ```
> <?php defined('BASEPATH') OR exit('No direct script access allowed'); ?>
>
> ERROR - 2024-05-05 21:31:16 --> [...]

@magicbug how to debug this error? There is no more information in th cloudlog logfile and in the mariadb error log.

magicbug, 2024-05-20:
Without knowing part of the codes triggering that error it will be impossible you'd need to check errors above and below that to see what part of the codes causing it

aaggiiss, 2024-05-20:
For my error -What do you need to know, I enter in the callsign and it writes that to the log. Gone back to paper logs meanwhile and would like to using it? Thanks Stuart

R1BLH, 2024-05-20:
@magicbug I could describe steps how to reproduce it. But I do not know how to find the SQL request generated by Cloudlog during new callsign lookup at qrz.com. Could you help me to find this SQL query?

Steps to reproduce:
1. open Add new QSO form
2. enter your own callsign (or a completely new callsign that has never been entered in your log, but has a page on qrz.com) in the Call field
3. fill in the rest of the QSO data.
Result -- no information is entered from qrz.com, error line in the log file.

aaggiiss, 2024-05-30:
I did an export and fresh install and reimported my data and it seems to work now .. as this has other comments on it other than mine and I am fixed. Not tried an update yet but not related.

closed 2024-05-30 by aaggiiss