magicbug/Cloudlog #3079: Feature Request: WAB Processing

aaggiiss, 2024-04-15:
Hi,

Just getting back on again after many years off and I have not found a good way to keep track of WAB workings etc.  Would it be possible to add something in like you have done for other squares and counties ?

Thanks

Stuart
G7JYF

magicbug, 2024-04-16:
Hi Stuart,

The answer is yes it would just need the code written to draw the wab squares which I don't have handy at the moment but once we have that it would be fairly straightforward.

Peter

magicbug, 2024-04-17:
@aaggiiss looked into this a little more and found a kml file with WAB squares I need to check how it functions before over committing but this feature does seem viable at least on first glance.

aaggiiss, 2024-04-17:
Great thank you :-)

magicbug, 2024-04-18:
[image]

Its starting to come together each square is clickable to have the table appear and it can highlight grids

Best way to handle the data will be using the SIG field so SIG name will be WAB and the sig info will be the square eg SP50

[image]

magicbug, 2024-04-20:
Now implimented fully in the dev branch under Awards -> Worked All Britain

closed 2024-04-20 by magicbug