magicbug/Cloudlog #3103: When LoTW is Down no message when manual sync is used

Reid-n0rc, 2024-05-15:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When lotw.arrl.org is down (as it is right now) and manual LoTW sync is used, nothing is displayed after the manual button is pressed.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Username -> Logbook of the world
2. Click on 'Manual Sync'
3. Nothing happens

**Expected behaviour**
There should be an indication that the process failed because lotw.arrl.org is not reachable.

stevendodd, 2024-05-15:
This is also happening for me, although I've not confirmed if logbook of the world is down or not

Reid-n0rc, 2024-05-15:
LoTW has been down all day.

Reid-n0rc, 2024-05-15:
LoTW has been down all day 
[image]

magicbug, 2024-05-15:
If the Cloudlog is unable to upload to LoTW it will fail silently if its down, any QSOs it tried to upload will be marked as not sent and it will just try it again.

You can see in /logbook that its the case by looking at the arrows for example from my logbook today

[image]

Like everyone else I wait for it to return :)

magicbug, 2024-05-15:
Worth noting added some extra code to say upload fails, however this info will only turn up on the screen when the PHP request times out, depending on how you setup your server this could take a while, you can check by running the URL used for the cronjob in the browser.

closed 2024-05-15 by magicbug