magicbug/Cloudlog #3105: Auto refresh home page

emics, 2024-05-16:
**This is an Enhance**
is great to auto refresh the home page, (the page opened clicking the cloud log logo in top left).

**Describe the Enhance**
When I use FT8 or other digital software is great to view log and map updated almost in real time!

73 IZ0IJD

magicbug, 2024-05-16:
In the dev branch I have made it so the logbook table reloads every 5secs.. the map itself needs a pretty big rewrite to support reloading every similar time period as a setInterval would create quite an annoying flicker as it is.

emics, 2024-05-16:
oh  great, thank you

magicbug, 2024-05-16:
At the moment when you turn on gridsquare map overlay it triggers a Uncaught TypeError: map.getBounds is not a function

closed 2024-05-20 by magicbug