magicbug/Cloudlog #3111: eQSL error: eQSL Nicknames in Station Profiles aren't defined

zcrowman, 2024-05-18:
I've been stuck trying to get a connection to eQSL. I've tried it with a nickname in eQSL and without a nickname. I reached out to eQSL about nicknames and he confirmed that indeed that I do have a nickname defined. But still no luck making a connection.

magicbug, 2024-05-18:
You have set username and password in the account page?

magicbug, 2024-05-18:
Other thing to try is enable logging in /application/config/config.php  and see what is in the application/logs

zcrowman, 2024-05-18:
> You have set username and password in the account page?

Are you referring to the eQSL username/password externally?

zcrowman, 2024-05-18:
I have added the user/pass to Cloudlog for eqsl.

zcrowman, 2024-05-18:
> Other thing to try is enable logging in /application/config/config.php and see what is in the application/logs

Okay. I've activated the log and don't see anything in the log to indicate an issue

zcrowman, 2024-06-15:
Issue resolved. I didn't realize that there was a place under station settings for this.

closed 2024-06-15 by zcrowman