magicbug/Cloudlog #3122: Filter by Mode issue in Advanced Log

novakeith, 2024-05-27:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When in the advanced log, and attempting to filter by mode, only the modes from the previous 250 QSOs are listed.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Advanced Logbook'
2. Click on 'Filters', select the 'Mode' selection box, and only the modes pulled from the previous 250 QSOs will appear. Ex. I have 300+ FT8 and FT4 QSOs as my most recent QSOs, so if I'm looking my older SSB QSOs, I do not have that option. 
3. issue occurs regardless of how many entries you choose to view ex. 250, 1000, etc..

**Expected behaviour**
Filter options should be extrapolated from the entirety of the logbook. 

**Screenshots**
[image]
Screenshot shows that I have Upper Sideband mode QSOs logged, but the filter results only show FT4 and FT8

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Linux (Debian 12)
 - Browser [e.g. chrome, safari] tested on Firefox 125.0.3

novakeith, 2024-05-27:
FYI This appears to be related to this PR #1566

magicbug, 2024-11-27:
now resolved in the dev branch