magicbug/Cloudlog #3136: Gridmap - When zoomed in to 6 char colouring isn't shown

magicbug, 2024-06-11:
### Discussed in https://github.com/magicbug/Cloudlog/discussions/3130

<div type='discussions-op-text'>

<sup>Originally posted by **emics** June  7, 2024</sup>
With v2.6.14 in Gridmap i can see all grid worked and confirmed (red and green square).
If I zoom in a 4 chars square i can see detailed grid (6 chars) but i don't have the square coloured green and red although my QSO are correct with 6 chars grid.

have you the same ?
thanks</div>

emics, 2024-06-11:
I Have investigated ,
the first thing is to enable: `$config['map_6digit_grids'] = TRUE;` in application/config/cloudlog.php
the second thing is wrong parameters in d3 variable (located in line 37 -  assets/js/leaflet/L.MaidenheadColouredGridMap.js)
```
redraw: function () {
		var d3 = new Array(20, 10, 10, 1, 1, 1, 1, 1, 1, 1, 1 / 24, 1 / 24, 1 / 24, 1 / 24, 1 / 24, 1 / 240, 1 / 240, 1 / 240, 1 / 240, 1 / 240 / 24, 1 / 240 / 24);
```
zomming the map some 6digit square are green and red but in the wrong position.
the bounds variable in line 67 contains the corners of confirmed square and the coordinates are wrong.

i hope to help someone to fix the bug!!!

magicbug, 2024-06-19:
Fixed

closed 2024-06-19 by magicbug

emics, 2024-06-27:
Hi Peter,
in the new [realease](https://github.com/magicbug/Cloudlog/releases/tag/2.6.15) there is always the problem in 4 digit gridsqare.
I show you some screen in GridSquare Map:

This is the grid of Rome (Italy)
[image]

if I zoom in the squares changes
[image]

another level of zoom and change again
[image]

the only QSO I have in JN61 grid are in
- JN61is
- JN61iu
- JN61gu [confirmed]
- JN61hv
- JN61fv

there is somthing wrong in d3 variable (located in line 37 - assets/js/leaflet/L.MaidenheadColouredGridMap.js)
read my comment [here](https://github.com/magicbug/Cloudlog/issues/3136#issuecomment-2161006216)

commit ccb9f592 (2024-06-19): [Gridsquare Map] Shows 2,4,6 grids

Fixes #3136