magicbug/Cloudlog #3144: eQSL Card Image Download / old cards are not downloaded

do4bmw, 2024-06-16:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
The problem occurs when I try to download my older eQSL cards. I have a bunch of cards in the download list, from August 4, 2020, to October 7, 2016. When I click on download, it shows me the error "The requested content cannot be loaded" ... newer cards can be downloaded without any problems.

Is there a way to ignore the old cards? Or to skip the download?

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'USER / eQSL Import/eQSL Card Image Download'
2. Click on 'Download un-synced eQSL cards'
3. See error in screeshot
[image]
[image]

do4bmw, 2024-06-16:
Version 2.6.14

and the problem is only for eQSL cards , older then 04.Aug 2020

do4bmw, 2024-06-16:
ok , one more info

its not that no card oder then 2020 will not be downloaded ... it looks like its just these in the screenshot
because when i "view eQSL Cards" there are lot of cards older then these above

see screenshot , these are all OK 

[image]

do4bmw, 2024-07-01:
anyone ?

do4bmw, 2024-07-18:
still not fixed / no solution

magicbug, 2024-09-27:
Its not so much no fix, this is a limitation of eQSL

closed 2024-09-27 by magicbug

do4bmw, 2024-09-27:
> Its not so much no fix, this is a limitation of eQSL

so there is also no chance to set these QSL Cards as allready downloaded in Cloudlog ? 
because this would help so much , becuase the script will not try these cards over and over again

a simple flag "Already Downloaded" would help