magicbug/Cloudlog #3158: Context menu only shown partially

Nailcoding, 2024-06-27:
In Logbook overview, when hovering over the edit button of a QSO, the context menu is shown only partially. Cut-off at the end of the line. 

Version: 2.6.15
Browser: Chrome
Hosted: Linux server at a hosting provider

emics, 2024-06-27:
can you give a screenshot and version affected.
tnx

Nailcoding, 2024-06-27:
[image]

emics, 2024-06-27:
hi @Nailcoding , while waiting for the release of the new release you can manually edit line 799 of the file [assets/css/general.css](https://github.com/magicbug/Cloudlog/commit/9d44e93b95e7a9cb64b192c56ad8fc1dc34ef708#diff-5e79dd776ecb12fe07475a1e0e8073a872a014fd554ec61309f573d483b78c07) with a text editor
as shown on this page from red to green
https://github.com/magicbug/Cloudlog/commit/9d44e93b95e7a9cb64b192c56ad8fc1dc34ef708

closed 2024-09-27 by magicbug