magicbug/Cloudlog #3161: QSOs uploaded via CL API do not appear in logbook

minorsecond, 2024-07-02:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When uploading QSOs via the CloudLog API, they do not appear in the main page, or in the logbook page, although the total QSO count changes. When re-entering a callsign in the Live or Post QSO entry page, the API-uploaded entries do appear. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Upload QSO via API
2. Click on Logbook -> Overview or Logbook -> advanced
3. See that the new QSOs are missing

**Expected behaviour**
The new contacts should appear in the logbook views.

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser Firefox
 - Firefox Version 127.0.1
 - Cloudlog Version 2.6.15

magicbug, 2024-07-02:
First thought would be are you sending them to the correct Station Location ID and is that ID linked to the current Station Logbook?

magicbug, 2024-08-13:
Closing due to no reply

closed 2024-08-13 by magicbug