magicbug/Cloudlog #3165: undefined flag on station location

thecocohead, 2024-07-09:
**Describe the bug**
When the user enables the station plot on the map, the flag shows up as `undefined`. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Ensure map view is enabled on logbook overview screen (Admin > Global Options > Appearance > Logbook Map > Enabled)
2. Enable station view on Logbook (Edit user > Cloudlog Preferences > Station > Display)
3. Go to logbook overview screen 
4. Click on station
5. Pop-up will say `undefined` where the flag should be

**Expected behaviour**
`undefined` should either be removed (showing nothing at all), or replaced with the station's DXCC entities' flag. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Windows 11
- Browser: Firefox
- Version: 127.0.2

**Smartphone (please complete the following information):**
 - Samsung Galaxy S23
 - OS: Android 14
 - Browser: Firefox
 - Version: 127.0.2

emics, 2024-07-11:
hi @thecocohead , do you have `undefined` in all stations popup, did you see some flag?
Have you try to enable flag on some table columns  (Edit user > Cloudlog Preferences > Logbook fields > Choose column 1 -> Flags)?

thecocohead, 2024-07-11:
@emics , I was able to fix this issue in PR #3167 which is currently merged into the dev branch. It was only shown for the user's station pop-up, not QSO stations.
I'm going to close this issue as completed for now.

closed 2024-07-11 by thecocohead

magicbug, 2024-07-11:
Thanks for the PR it was indeed a bug spotted it myself a couple of times!

PR #3167 (2024-07-11): Add station flag to map
Fixes #3165. Change pulls the active station's DXCC entity, gets the ISO code for it, then displays it alongside the station on the map popup. This uses very similar code to how flags are displayed for other map pins.

[image]