magicbug/Cloudlog #3189: QSO-export for all stationlogs not possible

dk5cr, 2024-09-10:
I tried to export all QSO from 9/1 to 9/10 this year for all stationprofiles. Due to few POTA-activities i show all QSO from all stationprofiles in my stationlog. 

So i tried to select the upper entry in the stationprofiles-dropdown for exporting all QSOs. 
Then the following website shows an error: 

http://192.xxx.xxx.xxx/cloudlog/index.php/adif/export_custom 
no file found or so. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to QSO export 
2. select first entry of stationprofiles (above the first real stationprofile) 
3. select from / to 
4. click export and see the error 

**Expected behaviour**
I'd like to see an ADIF-file with all exported QSO, referring to the stationprofiles shown in the active stationlog.

**Desktop (please complete the following information):**
 - OS: raspbian on RaspberryPi4 
 - Browser - doesn't matter, occurs on firefox, edge a.s.o. 

** Question/proposal: 
How can i export all QSO of all stationprofiles in the selected time period like shown in the CL overview?

dk5cr, 2024-09-26:
hello, 
is there anybody out there?

magicbug, 2024-09-26:
Currently you can only export one location at a time not per logbook

dk5cr, 2024-09-26:
Thanks for your reply. That's a pity, because i provided few POTA-stations, every with an own stationprofiles. 
No chance to export them only by date? Especially when they are grouped by the stationslog to be shown in the overview, it's a little strange, that they could not be exported in one click.

closed 2024-11-27 by magicbug