magicbug/Cloudlog #3195: Feature Request - add embedable QSO table code

zcrowman, 2024-09-23:
On a place like QRZ you can embed a table of your latest QSOs in your profile. You can also embed a table from Clublog. It would be nice to be able to do that with Cloudlog as well but with a bit more flexibility, ie number of latest QSOs to show, columns, etc. 

Thanks for the great work on this product. It has been great to use.

magicbug, 2024-09-27:
This is actually possible now although I don't think its documented anywhere if you use <URL to cloudlog>index.php/widgets/qsos/<station-logbook-public-slug>

remove the <> and fill with correct information it will give you a table that can be used inside an embed tag

petrkr, 2024-09-29:
> This is actually possible now although I don't think its documented anywhere if you use index.php/widgets/qsos/
> 
> remove the <> and fill with correct information it will give you a table that can be used inside an embed tag

github ate your < > things...

Put it to code block.

I using now using `visitor/<logname>` on mine QRZ page, but I can not put tag which will run scripts in iframe, so content is littlebit... bad

https://www.qrz.com/db/OK1PKR

zcrowman, 2024-09-29:
Unfortunately this doesn't seem applicable for those of us that have Cloudlog installed on our own servers and are self-hosting.

petrkr, 2024-09-29:
> Unfortunately this doesn't seem applicable for those of us that have Cloudlog installed on our own servers and are self-hosting.

If you have valid SSL certificate and public avaiilable domain, like me (https://log.ok1pkr.radio/) then it is not problem.

I meanwhile trial-error found that table. (https://log.ok1pkr.radio//widgets/qsos/OK1PKR)

It works, but it is too simple, more than for print, than for embedding somewhere.

Best will be take that visitors app and strip menu header and optionally (for example by some GET paramater) strip that right side-statistics.

also it could have by GET parameters customizable columns, but that would be too advanced Feature request already. For begin copy of visitors table without extra header/statistics would be exactly what I looking for. Of course without javascripts, so browser will NOT complain about cross-site-scripting

zcrowman, 2024-09-30:
I do have a valid SSL and publically available domain. But no qso table. I can see yours there but if I duplicate the URL for my website, nothing is there. Looking through the file structure there isn't a widgets folder that I could find.

petrkr, 2024-09-30:
> I do have a valid SSL and publically available domain. But no qso table. I can see yours there but if I duplicate the URL for my website, nothing is there. Looking through the file structure there isn't a widgets folder that I could find.

Of course if you use mine 'OK1PKR' it will NOT work for you. You have to use yours name.

Do not know your instance, maybe it is your callsign, maybe it's something like 'general' or 'default', whatever.

Maybe that widget is connected with that visitor's URL and you need create that first.

But as whole web is route driven, you will not find that folder, more likely you will find rewrite for that.

closed 2024-10-30 by magicbug