magicbug/Cloudlog #3201: Cabrillo needs to be adjusted per contest

petrkr, 2024-10-04:
For now cabrillo seems to have some hard-coded format. Which for example now is not for CQ WW RTTY contest enough.

Would be nice if there will be column definitions or sets or something like "default" and "custom", because this is for now first contest, which rejected data format.

Another contest was Czech OK contest, because here is one extra space, but that is also about bad parser on OK's page parser, because they need fixed width with exact number of chars. Which is NOT cabrillo format as specified here https://wwrof.org/cabrillo/cabrillo-qso-data/

[image]

petrkr, 2024-10-04:
This is example of accepted CQ WW RTTY cabrillo format:

```
START-OF-LOG: 3.0
CONTEST: CQ-WW-RTTY
CALLSIGN: OK1PKR
LOCATION: DX
CATEGORY-OPERATOR: SINGLE-OP
CATEGORY-ASSISTED: NON-ASSISTED
CATEGORY-BAND: ALL
CATEGORY-POWER: QRP
CATEGORY-MODE: RTTY
CATEGORY-TRANSMITTER: ONE
CATEGORY-STATION: FIXED
CATEGORY-OVERLAY: CLASSIC
GRID-LOCATOR: JO70FC
CLAIMED-SCORE: 
NAME: Petr Kracik
ADDRESS-CITY: 
ADDRESS-STATE-PROVINCE: 
ADDRESS-POSTALCODE: 
ADDRESS-COUNTRY: 
EMAIL: cqwwrtty@ok1pkr.radio
CREATED-BY: Cloudlog
QSO:  14093 RY 2024-09-29 1458 OK1PKR        599 15 DX F6PTT         599 14 DX 0
QSO:  28082 RY 2024-09-29 1551 OK1PKR        599 15 DX NA2NY         599 05 NY 0
END-OF-LOG:
```

So maybe it can be done if I will put "DX" word in exchange field

magicbug, 2024-10-30:
Cabrillo is best effort, ideally yes we need functions per contest for the layout but I really didn't want to make Cloudlog into a full blown contest logging system, there's far better desktop solutions available

petrkr, 2024-10-30:
> Cabrillo is best effort, ideally yes we need functions per contest for the layout but I really didn't want to make Cloudlog into a full blown contest logging system, there's far better desktop solutions available

Not really. for first it will split logs and you need to import/export them than just use one system.

and second is, not everyone use Windows and for example popular N1MM does not work under linux/mac or even Android / iOS tablet/phone

in opposite Cloudlog works everywhere and also support rigctl bridge. So it is multiplatform and multiusable. Everything is stored in ONE database, instead of those desktop solutions where every PC have own local database and then it is hard to merge it.

That is reason why I like, support and use Cloudlog.

closed 2024-11-27 by magicbug