magicbug/Cloudlog #3205: An uncaught Exception was encountered

2E0LXY, 2024-10-12:
Importing basic ADIF from Log4OM and get this error.

An uncaught Exception was encountered
Type: Error
Message: Call to undefined function mb_convert_encoding()
Filename: /var/www/html/application/libraries/Adif_parser.php
Line Number: 136

If I do a Logbook of the World - ADIF Import I get.

A PHP Error was encountered
Severity: 8192
Message: strtotime(): Passing null to parameter #1 ($datetime) of type string is deprecated
Filename: controllers/Lotw.php
Line Number: 726
An uncaught Exception was encountered
Type: Error
Message: Call to undefined function mb_convert_encoding()
Filename: /var/www/html/application/libraries/Adif_parser.php
Line Number: 136

magicbug, 2024-10-12:
Can you provide the adif file?

2E0LXY, 2024-10-13:
PHP Module mbstring wasn't installed.
Once installed all sorted, now to play with it!
Many thanks :-)

closed 2024-10-13 by 2E0LXY