magicbug/Cloudlog #3210: Unable to save QSO with 4U24OCT

kconkas, 2024-10-18:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
I am unable to save a QSO with station 4U24OCT (details in the screenshot). Clicking "Save QSO" seemingly does nothing - no errors, but the QSO itself isn't saved and the Post QSO  screen remains open. Other QSOs can be saved normally.

**To Reproduce**
Steps to reproduce the behaviour:
1. Click "QSO" -> "Post QSO"
2. Enter details as per below screenshot
3. Click "Save QSO"
4. Observe that nothing happens

**Expected behaviour**
QSO gets saved.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu Jammy Jellyfish
 - Browser Firefox
 - Version 131.0

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S24+
 - OS: Android 14
 - Browser Firefox, Chrome
 - Version 131.0.3, 129.0.6668.102

**Additional context**
Self-hosted instance, on commit 5dbe8b238c819812e5adf15c6a09d689e060b0c2, dating 8 October 2024.

magicbug, 2024-10-18:
I'm guessing the DXCC hasn't been populated, solution would be to manually set it or try updating the country files

kconkas, 2024-10-19:
Thanks for the prompt response @magicbug , country files update fixed it.

closed 2024-10-19 by kconkas