magicbug/Cloudlog #3221: No Frequency displayed

Jackdog44ron, 2024-10-31:
Every thing connects but unable to get frequency to show . So I log get random frequencies.

magicbug, 2024-10-31:
You're going to need to explain this a bit more, are you using Cloudlog CAT or just using the band selection?

Jackdog44ron, 2024-11-01:
Hi. I am using cloud cat and omnirig both ic7610 and ic7300 say connected as omnirig 1 and 2  it also says url is sync but I can add entries but get no band or frequency showing have to add it  manually RobSent from my Galaxy

magicbug, 2024-11-03:
- Is CloudlogCAT showing the rig data?
- In Cloudlog if you go to the your callsign dropdown top right and go to Hardware Interfaces do you see the radio data?
- Are you selecting the radio in the QSO panel under station radio

closed 2024-11-27 by magicbug