magicbug/Cloudlog #3227: Not all modes displayed in filter in advanced logbook

sebket, 2024-11-15:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
When using the "advanced logbook" and trying to filter after mode, not all modes are shown, in my case USB

**To Reproduce**
Go to Logbook, then Advanced
Click on Filters
Expand the dropdown list for Modes

**Expected behaviour**
I have QSOs in many different modes in my log, mostly various digimodes.
But there are also some QSOs in the Mode USB.
The mode USB is not shown and therefore not selectable.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Firefox
 - Version: 132.0.1

 - Version of Cloudlog: 2.6.16

magicbug, 2024-11-27:
This is functioning correctly it gets ALL the qsos from the locations in the logbook and it shows all the unique modes.

Unless you can provide an example of this not being the case

magicbug, 2024-11-27:
now resolved in the dev branch

closed 2024-11-27 by magicbug