magicbug/Cloudlog #3234: Error during install: cloudlog.options does not exist

john-terrell, 2024-12-03:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Installed fresh Ubuntu 22.04 LTS (also tried 24.04 LTS) with PHP7.4 (installed via PPA) and Apache 2 with CloudLog 2.6.16 (tried using the HEAD of 'master' as well).   After configuring the install via the install.php page, I receive the following error:

    Error Number: 1146
    Table 'cloudlog.options' doesn't exist
    SELECT * FROM `options` WHERE `autoload` = 'yes'
    Filename: models/Options_model.php
    Line Number: 14

Looking at the database, several tables have been created but no 'options' table exists.   I've tried this with a local MySQL Database as well as a remote one.   Same issue.

The only thing in the apache logs are:

[Tue Dec 03 22:34:01.381453 2024] [php7:warn] [pid 28743] [client 10.10.6.157:60135] PHP Warning:  unlink(/var/www/html/install/assets/install.sql): Permission denied in /var/www/html/install/index.php on line 17, referer: http://10.10.200.20/install/index.php
[Tue Dec 03 22:34:01.381610 2024] [php7:warn] [pid 28743] [client 10.10.6.157:60135] PHP Warning:  unlink(/var/www/html/install/config/config.php): Permission denied in /var/www/html/install/index.php on line 17, referer: http://10.10.200.20/install/index.php
[Tue Dec 03 22:34:01.381638 2024] [php7:warn] [pid 28743] [client 10.10.6.157:60135] PHP Warning:  unlink(/var/www/html/install/config/database.php): Permission denied in /var/www/html/install/index.php on line 17, referer: http://10.10.200.20/install/index.php
[Tue Dec 03 22:34:01.381704 2024] [php7:warn] [pid 28743] [client 10.10.6.157:60135] PHP Warning:  unlink(/var/www/html/install/includes/core_class.php): Permission denied in /var/www/html/install/index.php on line 17, referer: http://10.10.200.20/install/index.php
[Tue Dec 03 22:34:01.381724 2024] [php7:warn] [pid 28743] [client 10.10.6.157:60135] PHP Warning:  unlink(/var/www/html/install/includes/database_class.php): Permission denied in /var/www/html/install/index.php on line 17, referer: http://10.10.200.20/install/index.php
[Tue Dec 03 22:34:01.381741 2024] [php7:warn] [pid 28743] [client 10.10.6.157:60135] PHP Warning:  unlink(/var/www/html/install/index.php): Permission denied in /var/www/html/install/index.php on line 17, referer: http://10.10.200.20/install/index.php

    To see if these permissions errors were related, I set all of those files to RWX - re-ran the install script with the same result (but with the errors no longer present in the log.    Not sure why the install script is trying to remove those files...but it seems unrelated.)

I also ran the update_cloudlog.sh script as well but that didn't help.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser Safari and Chrome

john-terrell, 2024-12-04:
Ok, just tried again with a new install of OS and app.   Same thing.   Does the install do any logging I could investigate?

john-terrell, 2024-12-05:
@magicbug - On thing I tried was to run the install.sql directly on the database and it failed with this:

11:30:08	INSERT INTO `migrations` VALUES ('15')	Error Code: 1105. Percona-XtraDB-Cluster prohibits use of DML command on a table (cloudlog.migrations) without an explicit primary key with pxc_strict_mode = ENFORCING or MASTER	0.0065 sec

john-terrell, 2024-12-06:
@magicbug - The issue with the above error is the fact that the primary key for the options table isn't specified as part of the "CREATE TABLE" command.   Moving the primary key specification from the UPDATE command to the CREATE command fixes this.   I have a change in my fork that makes everything work (I also added a few missing "DROP TABLE IF EXISTS"' entries).    Let me know if you want me to submit a pull request.

magicbug, 2025-01-07:
Think I have this sorted now

closed 2025-01-07 by magicbug