magicbug/Cloudlog #3235: QRZ Download CPU Hogging.

ChrisL79, 2024-12-04:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
Everything has been working flawlessly for 7 months.

Cron job setup for downloading conformations from QRZ.

Last month or so, this makes CPU randomly hang at 50%..Sometimes after some time this will jump to 100%. The only way to stop this is to kill the PID or restart Apache.

**To Reproduce**

Enable cron for QRZ Download and wait. Could be hours or days.

**Expected behaviour**
For it to work without hogging the cpu.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Ubuntu 22.04.05 LTS.

I have updated Cloudlog to latest version and problem persists.

Is this an API issue on QRZ end?

magicbug, 2024-12-04:
To my knowledge nothing has changed in Cloudlog that could cause this, I'm guessing qrz change will need to investigate

ChrisL79, 2024-12-04:
Thanks. I had a feeling it would be on QRZ side as i rarely update Cloudlog. It had been sitting there for many months just doing its thing.

Posted to just see if anyone else was experiencing the same.

magicbug, 2025-04-18:
This is now resolved in dev branch

closed 2025-04-18 by magicbug