magicbug/Cloudlog #3244: Data too long for column 'user_lotw_password'

uiolee, 2024-12-27:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**

I use a string of length 128 as passwords for clublog and lotw.

When I set the `lotw` password in the cloudlog settings, cloudlog thrown an error. 
But when I set the `clublog` password, no error was thrown.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to cloudlog - account seting - Third Party Services
2. fill lotw username 
3. fill lotw password with a string of length 128. 
    for example: `W3uLmZfnFcwXnuuovFoggTFV3QJrSdMkgKAj7CmALfTAomLbmCzu9oQgxtY4AaxCRDRwzzDU9E4qU9VvK9JQV9ZQzic42CDbRJw4ffUyHtbsraMQWcpprq3FmoYDDJ33` 
5. Scroll down to bottom, save the changes.
6. See error

**Expected behaviour**

I expected that passwords of any length should be supported.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: windows 10
 - Browser: chrome
 - Version [e.g. 22]

```
Cloudlog Information
Version	2.6.16
Language	English
Base URL	http://cloudlog 
Migration	191
Server Information
Server Software	Apache/2.4.62 (Win64) PHP/8.2.27 mod_fcgid/2.3.10-dev
PHP Version	8.2.27
MySQL Version	11.5.2-MariaDB
Codeigniter
Version	3.2.0-dev
```
**Additional context**
Add any other context about the problem here.

magicbug, 2024-12-27:
I'll get that increased in the code base, until that happens you can manually change the columns properties in MariaDB

magicbug, 2025-01-07:
Now resolved in dev branch will make it into stable soon

commit 5bbeb096 (2025-03-05): Increase the lotw password size in the database

Fixes #3244
closed 2025-01-07 by magicbug