magicbug/Cloudlog #3246: PHP Error on Installation

lokiwho, 2024-12-27:
### Discussed in https://github.com/magicbug/Cloudlog/discussions/3245

<div type='discussions-op-text'>

<sup>Originally posted by **lokiwho** December 27, 2024</sup>
Howdy,

I have been over and over trying to install cloudlog outside a docker instance - right now I have apache on one server and mysql-server 8 on another, and I am attempting to have them talk to one another and I get a mysqli_sql_exception. I have tried this with nginx as well, incase it was a setting issue in the httpd, but I have not figured it out. Any ideas?

`[Fri Dec 27 17:42:33.631920 2024] [php:error] [pid 4633] [client 192.168.207.14:46324] PHP Fatal error:  Uncaught mysqli_sql_exception: Connection timed out in /var/www/cloudlog.html/install/includes/database_class.php:9\nStack trace:\n#0 /var/www/cloudlog/html/install/includes/database_class.php(9): mysqli->__construct()\n#1 /var/www/cloudlog/html/install/index.php(49): Database->create_database()\n#2 {main}\n  thrown in /var/www/cloudlog/html/install/includes/database_class.php on line 9, referer: https://[domainremoved'/install/
`

Thank you in advance.

73,
W1OKI
</div>

pa2eon, 2025-01-04:
What is the version of the Cloudlog source code you use .. I have en exception error also at version 2.6.9 ?
vy '73 - PA2EON

magicbug, 2025-01-07:
Connection timed out means your server is timing out, if you are first setting up Cloudlog setting up the dXCC tables can take a while you will need to adjust your server settings

closed 2025-01-07 by magicbug