magicbug/Cloudlog #3248: Strange behaviour of xOTA-fields in Cloudlog

dk5cr, 2025-01-06:
When entering a POTA-number (or other xOTA), i have to click in the field and type in the number, e.g. NO-2542. 
Afterwards cloudlog shows me kinda selection of matching numbers, in this case exactly the NO-2542, which i typed in few seconds ago. 
When working same POTA park next time, the number is NOT selectable from the list, list is empty at every time, so entered POTA number is not stored (as well as other xOTA-activities show the same behaviour). 
It seems to be a bug, that typed and selected number is not stored, otherwise i have to ask: Why typing in AND selecting the typed POTA-number?

**To Reproduce**
Steps to reproduce the behaviour:
1. add new QSO (doesn't matter if live or with time lag) 
2. change to 3. tab (in German: "Allgemeines") 
3. scroll down to POTA-referencenumber
4. type in a typical POTA-number as US-3004 or NO-2542
5. see, how selection list appears with the need to click again, even if content is already typed in.

**Expected behaviour**
I expected NO dropdown list but a normal text field to type in xOTA-numbers. Otherwise already worked xOTA-destinations should be given in the dropdown list as possible selection, according to the type letters, e.g. country selection like DE- or US- or NO-, only the countryspecific worked parks / references should appear. 

[image]

**Desktop (please complete the following information):**
 - OS Raspbian, current version 
 - Browser firefox, but same bevaiour on cellphone, linux, windows with Edge 
 - Version 133.0.3

magicbug, 2025-01-07:
While "Add" appears you should be able to tab out of the field without any issue?

dk5cr, 2025-01-07:
Yes, i can tap/click somewhere else on the screen except the POTA field itself . But after clicking elsewhere the typed value is not in the POTA field...