magicbug/Cloudlog #3255: QRZ Sync does not work

fabltd, 2025-02-12:
Hi 

The QRZ download QSO option does not work.  QSO's are downloaded to the UI but never added to the logbook. 

QSO confirmation also appears not to be working as intended.

API key is present.

Running V 2.6.16 on a Linux Ubuntu VM.

Thanks

magicbug, 2025-02-13:
Interesting, I don't use the QRZ Download function, but I'll get it setup again and double check its all functional

magicbug, 2025-02-14:
I've just tested this here so fresh upload to QRZ Logbook then download the matches and marked the QSO correctly

[image]

This is on my live setup 

```
Server Software	Apache
PHP Version	7.4.33
MySQL Version	8.0.41-0ubuntu0.20.04.1

```

magicbug, 2025-02-14:
I think the next step would be to enable development mode in `index.php` and edit `application/config/config.php` to enable logging of everything then try to do an import and check for any errors

magicbug, 2025-04-18:
The download code has been rewritten to handle bulk processing and is now in the dev branch seems to be functioning correctly

closed 2025-04-18 by magicbug