magicbug/Cloudlog #3263: QRZ Lookup and Persistent connections are disabled error

aaggiiss, 2025-03-15:
Hi, 

Just built myself a new install on my Rocky 8 Digital Ocean server as previously hosted on a vm on my laptop and I cannot get my QRZ lookup to work and the icons do not appear.

I have configured the config as per my local copy but the the icons do not appear and lookups fail

Server:

Version	2.6.17
Language	English
Server Software	Apache
PHP Version	8.2.25
MySQL Version	10.3.39-MariaDB

In Developer mode I get:

A PHP Error was encountered

Severity: Warning

Message: mysqli::real_connect(): Persistent connections are disabled. Downgrading to normal

Filename: mysqli/mysqli_driver.php

Line Number: 203

Production it goes away and the QRZ comes up saying  “Lookup not configured. Please review configuration.”

This 

curl -L 'http://xmldata.qrz.com/xml/current/?username=g7jyf;password=*******;callsign=g7jyf ‘

works on the server fine as I have full access.  

I have the config correct in the config.php, I have copied my password into qrz itself and it works fine

$config['qrz_username'] = "g7jyf";
$config['qrz_password'] = "********";

So I switched developer mode on to look at he error but only get the “ Persistent connections are disabled. Downgrading to normal” error but no either errors which may help with my qrz issue. 

Any suggestions ?

Many thanks

Stuart

magicbug, 2025-04-18:
I'm not 100% sure whats causing that you can update database.php in application/config but this almost feels like a PHP version issue

aaggiiss, 2025-04-18:
Hi, as requested - take it you meant uploaded than updated

$db['default'] = array(
        // The following values will probably need to be changed.
        'dsn'   => '',
        'hostname' => 'localhost',
        'username' => '<redacted>',
        'password' => '<redracted>',
        'database' => 'cloudlog',
        // The following values can probably stay the same.
        'dbdriver' => 'mysqli',
        'dbprefix' => '',
        'pconnect' => TRUE,
        'db_debug' => (ENVIRONMENT !== 'production'),
        'cache_on' => FALSE,
        'cachedir' => '',
        'char_set' => 'utf8mb4',
        'dbcollat' => 'utf8mb4_general_ci',
        'swap_pre' => '',
        'encrypt' => FALSE,
        'compress' => FALSE,
        'stricton' => FALSE,
        'failover' => array(),
        'save_queries' => TRUE

Thanks

Stuart

aaggiiss, 2025-05-18:
Further information required was added a month ago - any update ? Thanks