magicbug/Cloudlog #3344: Mode and Sub-Mode bug in Statistics screen (and maybe others)

g4klx, 2025-09-20:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the forum see readme for details**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

g4klx, 2025-09-20:
In the Statistics screens, and maybe others, when an option to filter by Mode is selected, and a mode which is a Sub-Mode is selected, the results are empty even though data exists. For example I have QSOs in FT4 and Q65, but these are not shown unless the Mode is set to All, but if I select something like CW then the relevant QSOs are found.

magicbug, 2025-09-20:
Thanks, been out today but this will get fixed tomorrow